veebch/photon #7: im missing something

Stefanvoor, 2022-12-08:
Hi, im pretty new to this sort of thing and running into an issue.
when i run de code in thonny i get an error, name 'ssd' isn't defined.
am i doing something wrong here or is there something im not seeing, being a novice at this?

>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 195, in <module>
  File "<stdin>", line 24, in splash
NameError: name 'ssd' isn't defined
>>>

veebch, 2022-12-09:
Welcome aboard :)

It looks like the file that defines ssd might not be loaded onto the pi? Is the OLED screen connected?

Stefanvoor, 2022-12-09:
Yes the Oled is connected, when i run gui\demos\aclock.py it seems to be working. i get a clock on the display when its running

veebch, 2022-12-09:
that gets ssid from a different place...... can you check that drivers/ssd1351/ssd1351_16bit.py is on the pico?

Stefanvoor, 2022-12-09:
they are indeed on the pico

veebch, 2022-12-09:
Looking at the code it looks like there is an issue with the sensor that means that you've found a bug.

Stefanvoor, 2022-12-10:
yes, its working perfectly over usb now, but when i power it on via a lipo its not running. the power light is on but the screen isnt coming up.

veebch, 2022-12-10:
Progress!

I've not encountered the problem with the lipo. I seem to recall that if you power it on via the lipo then you can still connect to thonny over usb. So you could run it interactively to get a better idea of the issue.

veebch, 2022-12-13:
Closing so you can add the specific LiPo issue, in the hope that someone can help

closed 2022-12-13 by veebch