veebch/photon #11: No display output

iamyd, 2023-01-21:
After assembled everything, there are no display. Then I thought it was issue with my solder, I re did all my soldering, still the same issue. I attempted to troubleshot using Thonny, the main.py ran without any error message. I decided to order a replacement screen, still nothing shows up. My multimeter shows the display is getting 3.3 volt, and the wiring connection is connect to the pico lipo. I attempted to write a hello world using the diver in the repo, still nothing shows up. Any recommendation on what to try next?

veebch, 2023-01-21:
That sounds annoying.

 Is the screen the exact same model? 

I would try to get it working on a breadboard while debugging.  

OLEDs look like they are doing nothing until an image is sent to them. Double check the gpio pins you're using.

iamyd, 2023-01-21:
It is the exactly same waveshare oled screen. I guess the bread board will be helpful to debug. 

The Pico is working too, the micro python works well. I checked again on the pin and even read the oled manual. 

In the instruction, it said CLK is connected to clock data input, but we connected to ground, does it means we don’t use clock?

veebch, 2023-01-21:
GPIO 18 on the diagram... not pin number 18 (which is gnd)

iamyd, 2023-01-21:
Issue found, thank you so much @veebch!

I connected the all GPIO pin to the regular pin. I am new to this DIY world, if would be helpful to also mark the pin number, for the pico pi.

closed 2023-01-21 by iamyd

veebch, 2023-01-21:
I've added the pin numbers to the OLED table in the readme. I didn't add them to all the tables because hopefully including them in one table only will make people see that they are two different notations. 

Well done on being so tenacious!