veebch/photon #12: Pico Lipo Battery Level Shows -185% when startup

iamyd, 2023-01-23:
When I am starting the Photon on the splash screen it will show -185% (Sometime -186%), however if I am reading from the Thonny it will display the correct percentage. When I am debugging from Thonny, it will shows the correct voltage on screen as well. Have you experience it before? 

I am trying to googling it, but it doesn't seems to go anywhere. In the main.py, the power management code is the same as the example pimoroni provided. Everything else works perfectly, what should I try to debug next?

veebch, 2023-01-23:
These are the few lines in the code that refers to that number

full_battery = 4.2                  # these are our reference voltages for a full/empty battery, in volts
empty_battery = 2.8                 # the values could vary by battery size/manufacturer so you might need to adjust them

voltage = vsys.read_u16() * conversion_factor
percentage = 100 * ((voltage - empty_battery) / (full_battery - empty_battery))

A negative value means that  Voltage < Empty battery. If that battery is at 100 percent then you just need to tweak the parameters a bit. If you've not changed anything then (Voltage - 2.8) / 1.4 = -1.86, so your battery is at 2.8 - 1.4 * 1.86 , about .2V? That see,s low, but my mental arithmetic may be off :)

veebch, 2023-01-23:
The other number that is important is conversion factor, maybe the Pico Lipo reads volatage differently to the shim that was originally used?

iamyd, 2023-01-24:
The conversion factor looks right, here is an example from pimoroni. If I try to debug this pico, the reading is also correct.

iamyd, 2023-01-24:
This is when I am trying to debug using Thonny. 
[image]

veebch, 2023-01-28:
so that would suggest first setting full _battery as **2.8** ((2/3)*4.2),  and change empty_battery to **1** and you'd want to make a note of the percentage just before the power dies and then use that value to determine how to adjust that to 0 percent by calculating the correct value of empty_batttery

There seems to be something amiss for that conversion factor though, the voltage reading when plugged in to usb should be 5v(?) or maybe 3.3v....... anyway, definitely not 2.8v, which is what it needs to be to give 67 percent.

iamyd, 2023-01-30:
If I meeting the full_battery to 2.8 and empty battery to 1, then the power level is -45%, which still doesn't seem right. You are right about the conversion factor, I think there might be an issue there. If I use 4 * 3.3 / 65535 as my conversion factor, I can get 100% when I am debugging from Thonny

iamyd, 2023-01-30:
If I use the conversion factor of 74.5 * 3.3 / 65535, my fully charged battery will shows 100%, according to some math calculations, the vsys.read_u16() is only reading 1320.63. If we assume the 4 * 3.3 / 65535 is the correct conversion factor (vsys.read_u16() read 24565 when plug into USB which should be 5 volt). That being said, my battery voltage is only 0.28 volt when fully charged which doesn't seems to be correct.

veebch, 2023-01-30:
Can you read the voltage directly off the battery using a multimeter?

iamyd, 2023-01-31:
It is 3.7 volt currently, a fellow redditor ([ToLoveSome](https://www.reddit.com/user/ToLoveSome) also having issue with the battery level. He recommended to read the voltage 10,000 times and for some reason it worked. 

I tested and confirmed it worked, then it get me to think when it is reading 10,000 times, it is just reading it again and again. Why don't we just let it wait for a period of time, I tested 0.1 sec all the way to 1 sec in an increment of 0.1 sec. I found that after 0.3 sec, the voltage display correctly, it shows 3.7 volts which is the spec of my battery.

After reading the battery specs, I found that the overcharge protection voltage is 4.28 V, and the overdischarge protection voltage is 3.0 V. I think the battery level need to be adjusted per battery.

veebch, 2023-02-01:
Merged.... closing :)

closed 2023-02-01 by veebch