veebch/photon #15: Rotate Display

polygonfuture, 2023-03-22:
Hi there,

I am building a custom case and pcb to house all the components in a different form factor.  Unfortuantely the design requires me to rotate the display 180.   I'm having trouble finding examples of how you'd rotate the display 180 with micro-python.  Have you looked into this before or have a solution?

veebch, 2023-03-22:
Not looked into it, but I would see if I could rotate the framebuffer 180 degrees. I'll look into it....

veebch, 2023-03-31:
Did you make any progress on this?

polygonfuture, 2023-04-18:
I solved the issue by sending a hardware command SET-REMAP and a corresponding 7bit byte command to the SSD1351 chipset.

In this case its two lines that are added to the ssd1351_16bit.py file just after we send these byte commands for initializing the SSD1351 in the` def __init__` section:

```
        # See above comment to explain this allocation-saving gibberish.
        self._write(b'\xfd\x12\xfd\xb1\xae\xb3\xf1\xca\x7f\xa0\x74'\
        b'\x15\x00\x7f\x75\x00\x7f\xa1\x00\xa2\x00\xb5\x00\xab\x01'\
        b'\xb1\x32\xbe\x05\xa6\xc1\xc8\x80\xc8\xc7\x0f'\
        b'\xb4\xa0\xb5\x55\xb6\x01\xaf', 0)
```

```
        self._write(b'\xA0',0)  #SET REMAP COMMAND
        self._write(b'\x76',1)  #SEND 7 BIT COMMAND IN HEX FORMAT THAT FLIPS DISPLAY.  IN BINARY:  01110110
```

First line tells the chipset we are going to make changes to the SET REMAP values.
Second line sets a 7bit value that changes 6 different settings on the chipset.   

There is value table I used to know what to set these to in [this datasheet ](https://www.waveshare.com/w/upload/a/a7/SSD1351-Revision_1.5.pdf)on page 32.   The A[1] bit is the value swaps the column address, and A[4] bit is how we swap the row / scan order.

veebch, 2023-04-18:
Woah.... that's lower level than I was thinking.

polygonfuture, 2023-04-18:
I tried for 5 minutes to flip the framebuffer in the show() command and gave up and just found that command in the datasheet.  haha