veebch/photon #16: BH1750 instead of BH1745

Mojojojooo, 2023-06-01:
Hi, I'm planning to build this and currently BH1745( Colour and luminance sensor) has been unavailable for a long time. Since the calculation part just shows that luminance is used. Can BH1750 which is just a luminance sensor be used instead of BH1745 ?

Mojojojooo, 2023-06-01:
How are the RGB values being used for calculation? As 1750 won't have RBG values any idea how can we bypass it ?

veebch, 2023-06-01:
They are only used in determining the colour of the EV reading. It should be easy to bypass.

veebch, 2023-06-05:
I would just set it to white (255,255,255)

polygonfuture, 2023-06-21:
@Mojojojooo BH1745 is not unavailable.  

Pimoroni keeps it in stock regularly:  https://shop.pimoroni.com/products/bh1745-luminance-and-colour-sensor-breakout?variant=12767599755347

Mojojojooo, 2023-06-21:
@polygonfuture I mean to say it's hard to get it in India. So I was looking for an alternative.

veebch, 2023-12-19:
No one chipped in, but I made an alternative. I'll share it as soon as it's ready

[preview](https://www.veeb.ch/projects/flash-aaahhh-ahhhhhh)

closed 2023-12-19 by veebch