veebch/photon #19: Don't work with me ;(

AndreScalaPT, 2023-08-11:
Hello, thank you for sharing your project with the community.
I don't see a discussion forum so I'll open as an issue, sorry.

[image]

So... I've followed everything... I believe.
So far, the LCD don't turn on and with the battery connected I don't have a signal in the board that is charging and don't turn on connected directily with the battery; the rest of the components I don't know how to test.

I used the pimoroni-picolipo_16mb-v1.20.4-micropython.uf2  because is the Pi that I have and uploaded the files using Thonny.

When I press the switch a light blink don't power up; if I have the Pi on and press the switch the power led only blinks once.

Help? 😅

Many thanks
🍻

veebch, 2023-08-12:
We've chatted about this on irc but I'll repeat myself here in case it's useful to anyone else. 

If the sensor has problems, it will have the behaviour you describe. Try running a simple example code from pimoroni to check that it is working.

veebch, 2023-08-12:
Closing :) Wires....always wires

closed 2023-08-12 by veebch

AndreScalaPT, 2023-08-12:
Yeess, yes, yes!
Just after soldering the cables to the sensor I was able to run everything.

Thanks @veebch 🍻