veebch/photon #27: installing to pico

illejandro, 2023-11-01:
i managed to get everything soldered and together, but i get no picture on the OLED and i also cant figure out how to install the necessary files to the pico. I have tried, Thonny, Putty, and Gitbash. Ive tried all codes provided and cant seem to figure it out

veebch, 2023-11-01:
Have you tried ampy? https://learn.adafruit.com/micropython-basics-load-files-and-run-code/install-ampy

illejandro, 2023-11-01:
I believe I installed it but got this. 
`Installing collected packages: pyserial, python-dotenv, colorama, click, adafruit-ampy
  WARNING: The scripts pyserial-miniterm.exe and pyserial-ports.exe are installed in 'C:\Users\User\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script dotenv.exe is installed in 'C:\Users\User\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script ampy.exe is installed in 'C:\Users\User\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed adafruit-ampy-1.1.0 click-8.1.7 colorama-0.4.6 pyserial-3.5 python-dotenv-1.0.0
`

Not sure if im missing something

veebch, 2023-11-01:
That looks ok, other than a few warnings about the PATH. If you can't run it, then that message has the path to to executable.

illejandro, 2023-11-01:
I think i installed it but now the ampy code just sits and doesnt show any status. I also triple checked my OLED and there is no image at all, only LED light on from liposhim

closed 2023-11-01 by illejandro

veebch, 2023-11-02:
Congratulations! It's *always* the wires :)