veebch/photon #30: Pico Lipo case : To support or not support ?

fitaine, 2024-05-20:
Hey there, this is a question for @040medien about the printing of the lipo case.

In the Readme.md there is a part about requirements which states "Make all parts printable without supports". When I checked the stl file before printing I saw a bridge in the middle of the case. 

[image]

It needs bridge so I was wondering : 

- It is a bug in the stl file : there is no support needed here.
or
- It is a compromise to get a better print: the case need support enabled.
I suppose the latter...

Thank you so much for that project, it will be a great help for me !

fitaine, 2024-05-24:
I ended up printing it with supports. Really nice case with some nice details !

040medien, 2024-05-24:
Thank you @fitaine - on my FDM printer small bridges like this don't need supports. That said, I don't remember what the bridge was for. I might have glued a component onto it, e.g. the battery.

040medien, 2024-08-08:
Thanks @veebch - I still use my photon regularly and proudly (even though I actually have an off-the-shelf light meter). Thank you for this project!

veebch, 2024-08-08:
Glad to hear it! I'm just making some updates to it (adding a new sensor that does flash and a battery compartment). I'm using your case as my starting point.