veebch/photon #33: Missing STL files?

dkallen78, 2024-11-18:
I'm looking for the right STL files to use and I can't find a proper contact method. I noticed the directory for the original case is missing the top and back parts of the print. And when looking at the directory for the larger case, the STL is missing the aperture for one of the switches. I have a Pico with a LiPo shim so I can't use the LiPo case, and I have the Pimoroni sensor, not your custom one so I can't use the Photon+

veebch, 2024-11-18:
It seems I just missed you on chat. The original case is in need of an update. It used cut acrylic on the front and back. It was really just a shell and the pico lipo case had some nice elements in there.

The photon+ stuff should be going into the repo in the next couple of weeks. The moment that is done, I'll update the original case.

I recall you asking about the lumisphere. Unless you want to recalibrate the code to the sensor behind the lumisphere then you can use without.

dkallen78, 2024-11-18:
no worries! is there any chance I can fit the Pico + shim and the original
Pimoroni sensor in the Photon+ enclosure?

Thanks!

veebch, 2024-11-19:
I used the same setup for the original so I can give it a whirl. At the very least it could form the basis of a better version that could be tweaked a little to fit everything in.

dkallen78, 2024-11-19:
That's going to be my first attempt, and i'll break out my mediocre blender
skills if I have to!

veebch, 2024-11-19:
That's the spirit! 

(feel free to add your improvements to the repository)